bitcraze/crazyflie-android-client #49: Idea: Wear app! (for CF 2.0)

AnotherKamila, 2015-02-01:
As Android Wear devices have BLE and an accelerometer, it should be possible to connect to the CF2.0 directly with the watch and control it using tilt for pitch/roll and a (single) joystick for thrust/yaw.

This would be useful for people like me who don't feel like flying with a 10'' tablet but don't have another BLE device besides the watch lying around. And it would be cool!

(I think that since the tilt control is already implemented, this would only require UI changes -- then the .apk could be `adb sideload`ed to the watch, in case making it available in the "standard" way is more work.)

fredg02, 2015-02-01:
Interesting thought. What would you use for controlling the thrust?

AnotherKamila, 2015-02-01:
Like I had described, thrust (and maybe also yaw) could be controlled on screen -- one joystick should be usable even on that small screen. And when the watch is held in hands rather than strapped to the wrist, it is easy to tilt it, so the watch's accelerometer could be used for directional control (pitch + roll).

I have no idea whether it would be convenient, but holding the watch in my hand and tilting it and moving my thumb around the screen isn't too awkward (actually, much less awkward than flying with a heavy 10'' tablet :D).

pappz, 2015-02-02:
I already implemented a solution to Pebble Smartwatch. https://github.com/pappz/crazyflie-android-client/tree/pebble

In that case for the thrust controlling I used the altitude hold mode on the Crazyflie. With the Pebble's buttons can raise the copter.

fredg02, 2018-03-06:
While it's an interesting idea, I won't implement it in the foreseeable future.

closed 2018-03-06 by fredg02