bitcraze/crazyflie-android-client #55: Custom Button Feature Request

knuckles904, 2015-08-26:
This is mentioned in comments for #33 but likely best added as a separate feature request. Most development effort focuses on primary projects like crazyflie firmware and python client. Nothing wrong with that, and it is likely due to amount of interest and bitcraze team doing a good job of making development on those projects easy to get into. 

Android (and I assume Iphone) apps are nice because most people always have a smartphone handy.  But apps lag behind in development (higher learning curve) and do not always allow new features available in the default python client, or support experimental firmware features. 

Instead of pulling development resources away from primary projects, I propose adding the following small but highly flexible features to the two smartphone apps:
1. Custom button(s)-Addition of one or multiple custom buttons to the app (disabled by default in settings, with some sort of "meant for advanced users/firmware development" disclaimer. Having a simple toggle or momentary button on screen which corresponds to a simple dummy Parameter in the firmware could allow anyone who's able to modify the firmware to route that "custom button 1" parameter to whatever feature they wanted to use/test/develop without having to dig into Android/Iphone app development. 
2. Simple console-A simple console (maybe another page in the settings menu) could allow mobile reading of any information from the CFlie by simply adding console prints of those params/log variables. If two way communication were implemented, this could also allow changing param values in the field. This would be a fine compromise feature instead of trying to replicate all the tabs from python client in each app. 

Closing statement: Adding in a few very small, simple, but highly flexible features into the app versions of the client will make it so app users still have possibility of benefiting from all the good continuing work done in CFlie firmware without requiring continual updates to the apps. 

Hope this is helpful. Thanks for all the hard work

fredg02, 2015-09-01:
Thanks for your suggestions.
Due to the nature of the necessary "ground work" to add two way communication it is currently only a small step between a "Just a custom button and a simple console"- and "the whole shebang"-solution.

Fortunately most of the "ground work" (aka Java based implementation of the cflib) is done. Altitude-hold mode is already working in a branch. Some refactoring quirks still need to be eliminated and tested because I don't want to release an app version that sends Crazyflies through the roof or smashes them into walls. ;)

I think, currently there are only a few parameters that make sense to be set from within the app (like altitude hold and LED-ring functions), but it may become more interesting when more "decks" will be released. The goal is definitely to use abstraction to avoid re-implementing every firmware feature.
A console is also on the short list.

fredg02, 2018-03-17:
A basic console view has been added in [release 0.7.0](https://github.com/bitcraze/crazyflie-android-client/releases/tag/v0.7.0).

There are currently no plans to add abstract "custom buttons" in the near future. Therefore I'm closing this issue with WON'T FIX.

closed 2018-03-17 by fredg02