bitcraze/crazyflie-android-client #63: Touch control behaves strange when yaw is negative

tobbeanton, 2017-01-05:
The problem seems to be related to yaw control only and when it is more then -100. Then the tough screen controls "flickers/jumps" and the yaw is reset to 0. If I limit the yaw to 99 the problem is gone.

Phone: Galaxy S6
Android: 6.0.1

fredg02, 2017-01-06:
I can't reproduce this bug with a Nexus 5x running Android 6.0.1. Can you reproduce it on any other phone?

krichardsson, 2017-01-09:
I have the same (or similar) problem on my Sony Xperia Z3 (D5803), Android 6.0.1.
For me it happens when the left touch pad/joystick (regardless of flight mode) is moved left further than -99. The application semi-freezes for a few seconds and then resizes the touch pads and makes them slightly smaller (!). After this they do not really work at all and the behavior is erratic.

commit 799dcef8 (2017-01-24): #63: Fix layout bug when Yaw is under -99

The bug comes from the YAW text wrapping when the number is
longer than 3 characters (-100). Fixed by increasing the
space allocated for the yaw display. Closes #63.
closed 2017-01-24 by ataffanel