bitcraze/crazyflie-android-client #64: Keep the app and the drone connected in background

Vortar, 2017-03-16:
I put a camera on the drone and watch the video on a webcam app on my phone with a Rx receiver plug on the usb port, but I can't pilot the drone and watch the video at the same time thanks to two different applications. I control the drone thanks to ps4 controller.
So when I minimize the crazyflie app, the drone is disconnected. So I want to know if it's possible to keep connected the crazyflie app in background (and keep the drone connected, when I use the gamepad) and use another app at the same time in front like webcam app to see in fpv? Or it is possible to modify the crazyflie app and add a video player and resolve the problem of connection in this case?

ataffanel, 2017-03-22:
I did some quick search and I am not sure it is possible to keep the gamepad input while being in the background. Do you have any insight on that @fredg02.

One solution is to use the split-screen capabilities of Android Nougat. I just tested and it works looking at a youtube video at the same time as controlling the Crazyflie with a bluetooth controller. You get only half of the screen but at least it works out of the box.

fredg02, 2017-03-27:
Sorry for the late reply. Like Arnaud said, there does not seem to be a way to capture gamepad input while an app is in the background.
I'd also recommend to try out the split-screen "workaround" for now.

closed 2017-03-27 by fredg02

qhelaine, 2017-03-28:
Hi !
Sorry for the late reply too. I think I will try this solution for now and
I will share you that I will reach to do. Thank you both. I'm really not an
expert in programation but if I find a program who
can keep working the crazyflie app in background, I will tell you.

Regards,

Quentin.