bitcraze/crazyflie-android-client #65: Bluetooth connection timeout if phone location disabled

ataffanel, 2017-04-13:
There has been one occurence where the bluetooth connection was always failing with a timeout if the android device location settings was disable: https://forum.bitcraze.io/viewtopic.php?f=2&t=2367&e=1&view=unread#p12015

In new android system bluetooth low energy is considered as a location technology so this makes sense but we should display a better error than connection timeout.

I haven't have time to confirm this on another device yet so, confirming the bug is the fist step.

fredg02, 2017-04-14:
Thanks for creating the bug, I will also try to reproduce the error. I have asked in the forum for a screenshot, because I'm not sure yet which exact setting is causing the error.

fredg02, 2017-04-15:
I cannot reproduce this error on a Nexus 5x with version 6.0.1 or 7.1.2. So I honestly don't know which "localization" or location settings is preventing the use of Bluetooth for the user.

fredg02, 2017-05-19:
After an extensive search and some tests this issue seems to be specific to the **HTC A9S**. There is currently no know way to reproduce this issue on another phone. Since a workaround has been found I have marked this as wontfix.

closed 2017-05-19 by fredg02