bitcraze/crazyflie-android-client #66: Improve information of firmware binary choice

krichardsson, 2017-06-14:
The firmware update mechanism gets a list of the binaries from the latest release in our github release repository. A list of binaries is displayed to enable the user to pick the correct one for the update. Currently it is not possible to understand which binary to use and this should be improved, possibly by displaying the file name in the list or something similar.

An example is the 2007.05 release that adds two lines to the list saying
"CF1 & CF2 2017.05"

fredg02, 2017-06-27:
Yes, different binaries (e.g. one with the Kalman estimator activated) are currently not covered. We need to find an easy file name scheme that allows to add infos like "kalman".
For example, ```crazyflie-<info>-YYYY.MM.zip```. WDYT?

We could add much more info in the manifest.json, but I'd like to create the release list drop down menu without downloading all release zip files first.

Side note: if the release notes text could be improved, this would automatically be reflected in the app (when you press the info button after selecting a firmware).

krichardsson, 2017-08-09:
Right now this problem is not acute as we have improved the firmware and do not have the need to release multiple binaries, but I think it would be nice to solve it for the future.

Agree, we don't want to download the zips to populate the drop down. I think your suggestion for a naming scheme looks good. To clarify, I would like to add the possibility of skipping the info since this is the format we have today, ```crazyflie[-info]-YYYY.MM.zip```

Examples:
* ```crazyflie-2018.02.zip```
* ```crazyflie-kalman-2018.02.zip```
* ```crazyflie-rocket-engine-support-2018.02.zip``` - can we use '-' in the info part?

ataffanel, 2017-08-09:
Yes we can use - in info part apparently, it [works with the regexp](https://regex101.com/r/NZOozj/1): 
```regexp
^[a-zA-Z0-9]+(?:-([a-zA-Z0-9\-]+))?-([0-9.]+)\.zip$
```

fredg02, 2017-08-10:
Thanks for the regexp Arnaud! :)

Unfortunately there will be a limit of characters for the info part (I need to check how many...), otherwise the drop down will overflow or mess up the layout.