bitcraze/crazyflie-android-client #87: Platform is displayed as "Unknown" in bootloader page

krichardsson, 2019-09-26:
When flashing a Crazyflie/Roadrunner from the bootloader page, the platform of the available releases is not displayed. Booth the CF2 and TAG releases appear as "Unknown 2019.09" which makes it hard for the user to understand which firmware flavour to pick. 

How to reproduce:
1. Click setting icon
1. Click Bootloader
1. Open drop down with firmware versions

Actual: 
The list contains
"Unknown 2019.09"
"Unknown 2019.09"
"Unknown 2019.02.1"
"Unknown 2019.02.1"
...

Expected:
"Tag 2019.09"
"Cf2 2019.09"
"Tag 2019.02.1"
"Cf2 2019.02.1"
...

fredg02, 2019-09-27:
Hey Kristoffer,
Thanks for reporting this. I'll take a look.

closed 2019-10-01 by fredg02

fredg02, 2019-10-01:
@krichardsson Should the tag firmwares even be shown in the Android client bootloader? Does it make sense to flash the tag from the Android client?

krichardsson, 2019-10-02:
Sweet! 
Not sure if the tag should be available or not. In theory it is possible to flash the Tag from the android client, but it is probably not the most common use case :-)
I suggest you leave it as it is.

commit 431f7934 (2019-10-01): Fix #87 - Platform is displayed as "Unknown" in bootloader page