bitcraze/crazyflie-android-client #93: Implement Assisted mode control.

Spordoz, 2021-07-02:
Implement a button that enables Altitude hold assist mode and the neccesary code to make it function.
Implement an preference for a command based control mode (left joystick replaced by 4 directional buttons : up, down, yaw right, yaw left) in combination with a takeoff command, kill command and land command.

fredg02, 2021-07-02:
Something similar was/is on my wish list, but I probably won't have time to work on this anytime soon. PRs are welcome! :)

Spordoz, 2021-07-02:
Cool, ill make a PR soon then. The only problem i am running into is that it works flawless with single packet sending, but fails over split packet. so right now it only works in combination with changes i made on the firmware side (including changing the parameter group name to have it fit in a single package). 
See https://github.com/Spordoz/crazyflie-firmware/tree/Altitude-Hold-baro-for-mobile for the firmware
and https://github.com/Spordoz/crazyflie-android-client/tree/Development for the client (i changed the branch name on githb desktop to be clearer, but it doesnt seem to have gone through online).

fredg02, 2021-07-02:
That sounds like a good motivation to fix the split packets. I'll try to spend some time on this in the coming days.